open-headset-interconnect-standard/ohis #8: Can standard contain alternative CW pin diagram

hcarter333, 2022-12-15:
Not so much an issue as a proposal. Could the standard also include a pinout for a CW keyer rather than a mic? Folks might like, (I suspect very much), to bring their own keyer as well.

I've built up a prototype of this idea at https://github.com/hcarter333/rockmite/issues/6

Thanks very much for putting together the standard! It inspired the work I did above. Also, if this isn't appropriate place for this question/suggestoin, please let me know, and I'll close or delete the issue. Thanks again! 
73 de KD0FNR

SmittyHalibut, 2022-12-15:
I literally just responded to your email to the OHIS mailing list and came here to enter an Issue in the backlog.  You beat me to it.  :-)

This is absolutely an appropriate place for this.  Thank you for submitting your idea.  It's a great one.

Unfortunately, the pinouts you listed above are not OHIS compliant.  OHIS is the definition of the signals, and what pins they go on.  Repurposing the Mic+ and Mic GND pins for contact closures breaks OHIS compatibility.

Instead, I propose the PTT line be extended from "Contact closure to ground for PTT" to a more complex analog signal:
* 1.00 * V_cc: Idle.
  * This is the "pulled up" state, same as before.
* 0.75 * V_cc: Button 3, Dit, Up
* 0.50 * V_cc: Button 2, Function
* 0.25 * V_cc: Button 1, Dah, Down
* 0.00 * V_cc (aka GND): PTT.
  * This is the "contact closed to ground" state, same as before.

This should probably be an optional extension that not every device has to support. eg: some radios that don't have or use extra buttons like a keyer, won't bother implementing these.

Encoding this would be pretty easy, just a voltage divider, so long as the pull-up impedance is known, or higher than the voltage dividers.

Decoding is more tricky.  Probably a pair of LM393 comparators, a voltage divider ladder, and some logic gates on the output to select only one output at a time.  Unless someone knows a better way to do this.

If I help with the circuits for this, would you be willing to prototype and test this @hcarter333 ?

hcarter333, 2022-12-18:
@SmittyHalibut I'd be will to prototype. The circuits as described above seem straightforward enough. Cool!

SmittyHalibut, 2023-02-27:
Updating summary to reflect how I'm thinking about this now.

SmittyHalibut, 2024-02-06:
This came up on the OHIS mailing list again: https://groups.io/g/ohis/topic/classic_radio_integration/102146840

Rather than define it by voltages as in https://groups.io/g/ohis/topic/classic_radio_integration/102146840, I think I like defining it by impedance instead.  Something like:
* Hi-Z is idle.
* 0R to ground is PTT.  (These two make it backward compatible with the current design.)
* 10k to ground is Dit.
* 33k to ground is Dah.
* 7.7k to ground is Dit and Dah together (spoiler: 10k||33k=7.7k)