open-headset-interconnect-standard/ohis #15: Microphone spec levels, generation, and calibration

gx1400, 2023-12-29:
The OHIS spec calls out

`The Microphone signal is a typical electret microphone signal: -45dBV +/- 3dbV into 600 to 1k
ohm`

Then goes on to describe the amplification needed for a dynamic mic

1. Should the spec section 3.3 be more clear about what an acceptable mic signal level is? Specifically " -48dBV to -42 dBV into a 1000 ohm resistive load" 

I wonder if later in the spec you could outline the intent and methods for achieving those levels, for example the reference circuitry.

2. Are these dBV measurements measured as Peak voltage or RMS?

3. Do you have any feedback on how to reliably generate audio into the mic to verify those levels? Or would you ultimately expect the end user to verify audio levels with signal reports and adjusting mic gain settings?

I think I noticed a mic level adjust pot on the Halibut user adapter product. I was curious how a "OHIS compliant/calibrated mic setup might vary from user to user based on talking loudness and mic handling (distance from mic, etc)

5. Have you considered a recommended calibration procedure for a given mic and user adapter? I thought it would be cool to have a reference circuit that has a OHIS mic input, a bnc input for an signal/tone generator for calibration, maybe a fixed load and an oscilloscope output. 

Then an end user could use a VU meter (even a few leds or an led bar segment, similar to an audio compressor meter) could help provide visual feedback that mic levels are appropriate for their hardware and talking style

gx1400, 2024-01-11:
1. User error on my side, but I do think there is some redundancy and conflicting information between section 3.3 and 5.3.  I think the spec is a little confusing in that 3.3 describes a 'typical' signal out into a load with a range of 600-1000 ohms.  I think Section 5.3 is again describing the impedance of the microphone element itself.  I wonder if your intent here was to provide contextual information, which is great but I think as a technical end user of a spec, it can muddy the "requirements" and lead to misinterpretation.

    ```
    3.3. Electrical Specification

    Microphone: The Microphone signal is a typical electret microphone signal: -45dBV +/- 3dbV into 600 to 1k
    ohm. The Radio device provides a DC bias voltage to power the microphone element, about 0.5mA across a
    2vDC microphone element (eg: 5vDC through 6.8k ohm resistor). Dynamic microphone elements require about
    10 to 15db of amplification, and for the DC bias to be blocked; this Standard provides a simple 5 component
    microphone pre-amp, powered by the DC bias, to bring a Dynamic microphone element up to Electret signal
    levels.
    ```

    I would suggest simplifying to:

    ```
    3.3 Electrical specification
    
    Microphone: The Microphone signal level shall be: -45dBV +/- 3dbV RMS into 600 ohm. 
    
    This level is typical of electet microphones. See section 7.1 for notes on amplifying Dynamic element microphones.
    ```

    Or even better, I might suggest to nix Section 3.3 completely and allow for 5.3 to be the sole place you give actual numbers.  Having it in multiple places can be confusing in a technical document.
<br>
2. Thanks for the consideration!
<br>
3. Nix this comment.  Good feedback, entirely dependent on user and their setup.  Per the VU meter, I'll comment over on [#16](https://github.com/open-headset-interconnect-standard/ohis/issues/16)
<br>
4. Sorry auto formatting and retyping sections in markdown is messy, I believe I typed this on my phone.
<br>
5. No additional comments!

SmittyHalibut, 2024-01-11:
1. Finally back at my computer not my phone, I actually reviewed 3.3 and 5.3, and I agree with you.  I'll greatly simplify Section 3.3 (not just Microphone) to be just a very quick summary, and point to 5.3 for full details.  However, I will include the DC bias voltage in the summary, not just the signal level.  But I won't go into details, I'll just say that its needed.
4. I never pass up an opportunity to throw in a good Monty Python quote. :-)