open-headset-interconnect-standard/ohis #16: Design an Open Source in-line volume meter

SmittyHalibut, 2024-01-10:
The idea is that this device is a piece of test equipment, not needed for full time use.  A club could have one for everyone to use to help them adjust their mic levels.  Then, once adjusted, remove it from the circuit.

* Has two RJ45 sockets in parallel.  It's just a straight passthrough.
* Uses power from Vcc and Power ground.
* Has a VU meter on the Mic level, with fixed set points.
* Has VU meters on Left and Right headphones, with fixed set points.
* Include a power LED, maybe with a volt meter (which is basically a DC VU meter).
* Include a PTT LED.  Don't add an appreciable load to the PTT line, though.  So control a MOSFET, or a comparator, or something similarly high impedance input.  Don't just drive the LED off the PTT line directly.
* Bonus points: Test points to measure: Mic/MicGnd, HP:L/HP:G, HP:R/HP:G, Vcc/Power Gnd/PTT

gx1400, 2024-01-11:
Love this, thanks for the consideration.

I think VU meter for headphones would be unnecessary, since those are are going to be set by the end user with a volume knob regardless.

While having this in-line does provide power, I was envisioning this to be a stand-alone device to test the headset side of the exchange exclusively.  Including a 5.5 x 2.1 mm power jack would allow it to be standalone without touching a radio.

Suggest:

- mic is driven into a fixed resistive load (600 ohm? 1k ohm?) to set a consistent level.
- I like the power and PTT indicators.
- I could see the benefit of a cheap tone generator (even something as simple as an 555 timer with adjustable level just to verify headphones are wired correctly on the user adapter and functional).  Maybe two tones, one for left, one for right.  Bonus feature, add a pot to adjust the tone freq for left and right independent.
- thru-holes for an SMA or BNC connector for Mics, and HPs to simplify hooking up to a scope

SmittyHalibut, 2024-01-11:
Just for kicks and giggles, here's that board:
[image]

User adapter connects on the left, Radio adapter on the right.  Large pushbutton 8P2T switches next to both ports to select whether that port is connected to local test equipment, or through to the other port.  Then all the visibility stuff in the middle: VU meters, BNC taps, some switches to select what a given BNC is measuring, a local PTT, etc.

gx1400, 2024-01-11:
Opportunity for reducing real estate: use edge mounted SMA connectors.  

[image]

I have about six other projects hanging that I need to close out before starting something new, but if you keep this ticket open, or open another issue for a standalone piece of headset test equipment.  I can try to start a KiCad circuit with some ideas to prototype in the future

SmittyHalibut, 2024-01-11:
RP-SMA!! Eww!  ;-)

Believe me, I know all about edge mount SMAs.  I literally have about 1000 in stock right now for various projects.  :-)

Since most o-scopes are BNC, I figured BNC was the right connector for this board.

As for you developing something, by all means!  Please do!  What's the whole-repo version of "I accept pull requests?" ;-)